habibrehmansg/infopanel #4: Profile asset folder becomes read only and cant save / update profile

centrdev, 2024-11-11:
Hey, 

Was putting together a profile, added the gfx and items, clicked on save and all working ok so far.
Restarted computer, brought up the profile and all but the bg image was gone and not saved 🤷‍♂️

Took a look in the folders and the profile asset folder is read only.
I then made it writable and made some updates to the profile and it the saved it correctly.
But the folder was then set as read only again...

Is this something you could look into?

habibrehmansg, 2024-11-11:
It’s probably windows defender and folder protection. Is that something you can rule out?

centrdev, 2024-11-11:
Hey @habibrehmansg , i did some more testing and i figured out what is going on.

Creating a profile, i add my gfx and items, but i did not click on "save" in the design and "panel items" pane. 
Doing this saves the most current setup, but if you forget to click on that save it will not save your "latest" changes.
Clicking on save on the profile details will also not save the latest changes of the items.

I'm going to admit I might be a dumbass, but i really think there should be a message / popup / dialog moving away from the design screen asking you to save you changes.

Some other suggestions would be to
- add an option for autosave every N minutes,
- make the save button on the profile page also save the changes done on design pane.

Hope this clears this out and would like to see some changes to the save 🫡

Keep up the great work

centrdev, 2024-11-12:
Yeah, i get that with the auto save, and it makes sense. But it would be great if it at least was an option, could be disabled by default. 

Yes, that would be great, could be a "(Unsaved changes)" label next to the save button or something, and maybe a popup message with a save button if you switch away from the design pane with unsaved changes?

You'll figure something out 🫡

habibrehmansg, 2024-12-06:
Closing this issue for now.

closed 2024-12-06 by habibrehmansg