habibrehmansg/infopanel #16: Turzx 5" Display doesn't initialize with InfoPanel after waking from sleep

adamckbatchelor, 2025-01-04:
As the title suggests, my Turzx 5" panel sometimes sync with InfoPanel after waking from sleep.  It normally does after reboot.  Let me know if I can provide any additional information.

adamckbatchelor, 2025-06-27:
hey @habibrehmansg  i just ran the latest main branch build and can confirm, my turzx panel boots and syncs after the computer wakes from sleep.  thanks for the fix!  will this be pushed to a main release anytime soon?

habibrehmansg, 2025-06-27:
> hey [@habibrehmansg](https://github.com/habibrehmansg) i just ran the latest main branch build and can confirm, my turzx panel boots and syncs after the computer wakes from sleep. thanks for the fix! will this be pushed to a main release anytime soon?

Thanks good news! Yes almost there now. Just monitoring for any other bugs. 1.3 is a major release and overhaul of multiple key features.

closed 2025-06-27 by habibrehmansg