habibrehmansg/infopanel #28: MSI Afterburner

xgrommx, 2025-03-10:
Hello! Could it be used with MSI Afterburner? Maybe exists some approach to catch data from MSI Afterburner when it will be turn on.

centrdev, 2025-03-10:
Hey, what data would like from afterburner? I'm assuming fps etc? If that's the case I got 2 different plugins, one "simple" and one more advanced coming soon.

Other gpu info you can already get from librehardwaremonitor.

F3NN3X, 2025-03-10:
Hey, what data would like from afterburner? I'm assuming fps etc? If that's the case I got 2 different plugins, one "simple" and one more advanced coming soon.

Other gpu info you can already get from librehardwaremonitor.

xgrommx, 2025-03-11:
which plugins?

F3NN3X, 2025-03-12:
with the upcoming release of InfoPanel there is added plugin support, and i have made 2 different FPS plugins;
one simple, that only display FPS, 1% low and frametime: https://github.com/F3NN3X/InfoPanel.FPS

and one more advanced using Intel Presentmon to display more details: https://github.com/F3NN3X/InfoPanel.IPFPS

the more advanced one runs presentmon in background and fetches data directly from presentmon.

habibrehmansg, 2025-03-19:
Closing this as it seems there are alternatives.