habibrehmansg/infopanel #38: Feature Request - Support for iCUE SDK

neowara, 2025-04-23:
Hey, I have been trying without luck to get my temp sensor inside of my Corsair Commander XT to display on my InfoPanel.
I have tried to get it working through HWiNFO but every time HWiNFO polls the sensor to get the data it will make all fans connected to that Commander XT to blink, if I change the polling time to something crazy I can make the blinking happen less often but is still there. There has to be a better way. It should be possible to get the data from the Commander XT directly, Corsair has an SDK that lets other apps control and poll directly from iCUE so that could be way forward, if InfoPanel could talk to iCUE itself it would be perfect. 

I know it's a very unique problem to have but I think the fix shouldn't be too hard to implement, I might even try to do it myself when I have some time. I have my whole custom water loop controlled through Corsair Commanders to manage the pump and fans speeds and lighting so I have no way around Corsair and their horrible software. If someone can point me to the right direction I might be able to make a PR eventually.

habibrehmansg, 2025-05-11:
The best way to handle this would be to write a plugin.

habibrehmansg, 2025-05-11:
Take a look at https://github.com/F3NN3X/InfoPanel.Spotify for a sample implementation, or https://github.com/habibrehmansg/infopanel/tree/main/InfoPanel.Plugins

Unfortunately, I haven't gotten round to writing the plugin documentation yet..