habibrehmansg/infopanel #42: Bug: Bar sensor item min range not working as expected

wojciechwojcik, 2025-05-20:
Recreate steps:

1. Create new bar with Min: 0, Max 40, assign a sensor item with current value around 25C
[image]

The bar is rendered correctly:
[image]

2. Change the  Min value to 20

[image]

The bar always shows full despite the value 25C not being anywhere near 40 max:
[image]

It seems that bar is calculated simply as
`Current Value / (Max-Min)`

which means 25C / (40 - 20) -> bar is 100% full.
which is not correct.

Instead it should be something like
`(Current Value - Min) / (Max-Min)`

So in this case:
`(25 - 20) / (40 - 20) = 5/20` -> bar should be 25% full.

@habibrehmansg please let me know if my analysis is correct here. This look like relatively simple fix. Thanks.

habibrehmansg, 2025-05-22:
@wojciechwojcik You are right. Thank you for the detailed explanation and screenshots. I've pushed a fix.

closed 2025-05-22 by habibrehmansg

wojciechwojcik, 2025-05-25:
Verified with https://github.com/habibrehmansg/infopanel/actions/runs/15184358385/artifacts/3176023636
Everything works as expected now.
Thanks @habibrehmansg