habibrehmansg/infopanel #50: New version of Turing 8.8 display 1.2 not compatible.

Oscar-Rod, 2025-06-03:
I just bought the display from the official store https://turzxmonitor.com/products/8-8inch-22-35cm-smart-screen?variant=49081647104347

I get the error "Display is not running. Check your COM port settings and try again"

I opened it, and it looks like it is a new version

[image]

habibrehmansg, 2025-06-04:
https://github.com/habibrehmansg/infopanel/actions/runs/15421099768

Can you try this build and see if the usb option for 1.1 works?

habibrehmansg, 2025-06-04:
> > https://github.com/habibrehmansg/infopanel/actions/runs/15421099768
> > Can you try this build and see if the usb option for 1.1 works?
> 
> It does not work, same error

I confirmed too that the new version seems to have a new encryption key and new methods. Unfortunately this won’t be fixed for a while.

habibrehmansg, 2025-06-05:
> > https://github.com/habibrehmansg/infopanel/actions/runs/15421099768
> > Can you try this build and see if the usb option for 1.1 works?
> 
> It does not work, same error

Are you willing to hop on to our discord and PM me?

habibrehmansg, 2025-06-06:
Closing this issue as I saw that @Oscar-Rod got it working on another forum with the USB option - which is already supported by IP, and have yet to receive any feedback.

closed 2025-06-06 by habibrehmansg