habibrehmansg/infopanel #58: Libre sensors

tonythemuss, 2025-06-28:
Hi, Just discovered this app and loving creating various panels for my case LCD. Everything seems to be working great using HWinfo sensors but I would prefer to use Libre after looking at the prices to remove the 12hr connection limit.

Issue I have is Libre is picking up all the sensors I need in the app but heaps have no data coming through, especially for my CPU. It will pick up usage but not temps or wattage for instance.

I downloaded and ran Libre standalone to see if it was actually capable of reading everything and it was, picks up all sensors I need and shows the correct values.

What could be the cause of it failing inside the infopanel app?

habibrehmansg, 2025-06-29:
It’s not failing. You probably have game mode enabled.

tonythemuss, 2025-06-29:
That was the issue. Thanks for the heads up. I enabled this when I originally downloaded and promptly forgot it was an option since I was using HWinfo with no issues.

closed 2025-06-29 by tonythemuss