habibrehmansg/infopanel #65: InfoPanel.sys detected as Trojan

DarthRogue, 2025-09-04:
MS Defender version 1.435.575.0 is giving a false positive detection identifying the InfoPanel.sys driver as being a Trojan:Win32/Vigorf.A.  You may want to consider working with MS to get this whitelisted and/or signed to prevent such issues in the future.  I already submitted a false positive notification to them, but wanted to make you aware nonetheless.  

Thanks for making this awesome tool!!

Sandokan71, 2025-09-06:
It's not actually a false positive.

It seems that InfoPanel and many other similar software programs, such as FanControl and many others from well-known brands, use an old driver that is no longer updated and has vulnerabilities.
So the problem is not so much InfoPanel and these software programs, but the vulnerability of the driver.
There are two options: ignore the Windows Defender warning and keep the vulnerability, or give up these software programs, at least temporarily.           

Translated with DeepL.com (free version)

MrPink777, 2025-09-14:
Thank you so much for this excellent software! 
I only use Libre. The infopanel.sys was quarantined. I restored it and added the infopanel folder to the exception list. Since then, reading the CPU temperature/clock/fan speed no longer works. The motherboard is also no longer displayed in Libre. I've reinstalled Infopanel several times, unfortunately without success.

closed 2025-10-15 by habibrehmansg